bportaluri/ALA #3: Compilation errors

Pramuspl, 2015-06-24:
When I'm trying to launch RgbLedSequence example I'm getting compilation errors:

RgbLedSequence.ino:22:28: error: 'colors_' was not declared in this scope

RgbLedSequence.ino:35:1: error: cannot convert 'AlaPalette' to 'AlaColor*' in initialization

RgbLedSequence.ino:35:1: error: cannot convert 'AlaPalette' to 'AlaColor*' in initialization

RgbLedSequence.ino:35:1: error: cannot convert 'AlaPalette' to 'AlaColor*' in initialization

RgbLedSequence.ino:35:1: error: cannot convert 'AlaPalette' to 'AlaColor*' in initialization

Pramuspl, 2015-06-24:
I found out that the format of the sequence is wrong (array of colours should not be last) so I changed the seq[] content to:
{
  { ALA_FADECOLORSLOOP, 10000, mycolors, 10000},
  { ALA_ENDSEQ, 0, 0, NULL }
};

 I also changed the initiated colours to:
AlaColor mycolors_[2] = { 0xFFFFFF, 0x000000 };
AlaPalette mycolors = { 2, mycolors_ };

The trouble I'm having now is that instead of going white-black-white, the only transition I get is red-green-red. Pins are set correctly (I checked all 9 possibilities and in none of the cases I can get white-red-white, it changes to blue-red-blue instead etc.) and no matter what values I put in mycolors_ array, the red-green-red sequence stays the same.

bportaluri, 2015-06-29:
I have fixed the compilation errors in the examples.
I have checked the example with both common cathode and common anode RGB LED and works fine.
Please download the latest version and let me know.
This is the code i have used.

```
#include "AlaLedRgb.h"

#define REDPIN 11
#define GREENPIN 9
#define BLUEPIN 10

AlaLedRgb rgbLed;

// custom palette :     black   , white   , black   , red     , green   , blue
AlaColor mycolors_[6] = { 0x000000, 0xFFFFFF, 0x000000, 0xFF0000, 0x00FF00, 0x0000FF };
AlaPalette mycolors = { 6, mycolors_ };

AlaSeq seq[] =
{
  { ALA_CYCLECOLORS, 3000, alaPalRgb, 3000 },
  { ALA_OFF, 1000, alaPalNull, 1000 },
  { ALA_ENDSEQ, 0, alaPalNull, 0 }
};

void setup()
{
  // initialize the RGB RED
  rgbLed.initPWM(REDPIN, GREENPIN, BLUEPIN);
  
  // set the brightness of the LED calibrating white
  rgbLed.setBrightness(0x338822);
  
  rgbLed.setAnimation(seq);
}

void loop()
{
  rgbLed.runAnimation();
}
```

Pramuspl, 2015-06-30:
How do you run this example with a common anode diode and get proper colours?

Pramuspl, 2015-07-13:
@bportaluri , have you checked what could be the problem?

bportaluri, 2015-10-25:
I have changed the order of variables in the AleSeq struct and modified the examples accordingly.
AlaColor is now the last one to allow lazy initialization of sequence if no palette is required.
Check out latest version of the examples in release 2.3.

closed 2017-07-30 by bportaluri