bportaluri/ALA #4: Using NeoPixel Library and ALA Library in the same sketch

tristandoo, 2015-10-14:
When using both libraries upon compiling the following errors occur:

Arduino: 1.6.5 (Windows 8.1), Board: "Arduino Uno"

ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:38: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::~Adafruit_NeoPixel()'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::~Adafruit_NeoPixel()'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::begin()'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::show()'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::setPin(unsigned char)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::setPixelColor(unsigned int, unsigned char, unsigned char, unsigned char)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::setPixelColor(unsigned int, unsigned long)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::Color(unsigned char, unsigned char, unsigned char)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::getPixelColor(unsigned int) const'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::getPixels() const'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::numPixels() const'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::setBrightness(unsigned char)'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::getBrightness() const'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
ALA\ExtNeoPixel.cpp.o: In function `Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned char, unsigned char)':
C:\Users\School\Documents\Arduino\libraries\ALA-master\src/ExtNeoPixel.cpp:36: multiple definition of `Adafruit_NeoPixel::clear()'
Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o:C:\Users\School\Documents\Arduino\libraries\Adafruit_NeoPixel-master/Adafruit_NeoPixel.cpp:51: first defined here
collect2.exe: error: ld returned 1 exit status

Error compiling.

Tried tweaking the .cpp file my self, but I never got it to compile without errors.

bportaluri, 2015-10-15:
I don't think this is possible because ALA is using NeoPixel library
internally so there are many conflicts.
If you need to access NeoPixel directly you can try to use the embedded one
instead add including another one.

tristandoo, 2015-10-15:
How do you suggest to do that?