bportaluri/ALA #9: Ram footprint

Testato, 2015-12-28:
i compiled the SimpleRgbLed on an UNO and it use 22KB of flash memory.

Is it absolutely necessary all this space ?
On .cpp i view that you include neopixel code and tc code, but if an user need only use a single simple RGB led he do not need neopixel or other code.

Maybe it is possible write some ifndef for dinamically reduce code usage ?

bportaluri, 2015-12-28:
Hi,
program space is consumed only by 10% from neopixel or TLC libraries.
The animation functions are consuming it. Since they are all referenced in the AlaLedRgb::setAnimationFunc method they are all included.
A possible solution is to use function pointers. Which may make the library less easier to use.
I will think about it. Thank you for the feedback.

eeik, 2018-03-23:
For what it's worth for others looking to lighten the load of the ALA Library if teetering on the edge of having your sketch fit on the Arduino or not, I had some decent luck in commenting out the animation functions within both the **AlaLed::setAnimationFunc** and **AlaLedRgb::setAnimationFunc** functions when I had decided that I would definitely not be using some of the available options.

[image]