bportaluri/ALA #11: Adafruit Trinket

Ham84, 2016-05-18:
Sorry I'm a Noob to most of this please bear with me.
I was trying to use this Library with the [Adafruit trinket](https://www.adafruit.com/product/1500).
But I get and #error "Unknown Chip!". I figured out this is because the chip on the trinket is not included In the [Library](https://github.com/bportaluri/ALA/blob/master/src/pinouts/chip_includes.h). My question is would/could support for the trinket be add?

bportaluri, 2016-05-19:
Sorry I don't have a Trinket board so I don't know how to test it.

closed 2016-05-19 by bportaluri

Ham84, 2016-05-20:
@bportaluri, If you want email an address lucasham84@hotmail.com and i will be more than happy to send you a 3.3v and 5v Trinket free.

bportaluri, 2016-05-20:
The problem is related to the TLC5940 library that doesn't support the Trinket.
I can try to introduce a #define to optionally disable it.