bportaluri/ALA #15: Teensy 3.2 Support

chrisgervang, 2017-09-22:
Do you think it's relatively straightforward to add support for the Teensy 3.2?

I see this library supports Teensy 2.0 and Teensy ++2.0 already. What did you need to know about the boards in order to add support for those? Perhaps I can add support for 3.2 if I knew more about what you did with the other boards. Thanks!

BNNorman, 2017-09-22:
According to this site: https://forum.pjrc.com/threads/22156-Uisng-ifdef-to-differentiate-between-teensy-2-0-teensy-2-0-and-teensy-3-0

For a teensy 3.0 you need,apparently, to check for this ..

#if defined(__MK20DX128__)

At the bottom of the webpage above there is a define for 3.1. (__MK20DX256__). I have no idea about a 3.2 but the 256 and 128 seem to relate to FLASH size.

I would imagine you then create a header teensy3.h similar to the other teensy headers and add an entry into the chip_includes.h

Now, this is supposition because I don't have a teensy - but I see no-one has helped you so far. You will need to get the specs and workout the correct contents for your .h file. 

That's how I would get started.

Hope that helps, apologies if not.

Brian

Mfink30, 2019-12-23:
This is my first post on GIT and I know this is an attempt to revive an old thread but I have been trying to work out how to add Teensy 3.5 support to this library and have not had any luck.   I went the route @BNNorman suggested but am having issues correctly creating a new header file by modifying "Teensypp_xxx6.h" to define the correct pins on the 3.5.  

Below is the pinout for the 3.5 which has mutliple of some of the pins and none of some of the others (at least as far as I can tell).  Any input will be greatly appreciated.

Thanks in advance,
Fink

[image]