bportaluri/ALA #16: Start sequence over?

Cue777, 2017-11-29:
Im running the rgbstripsequence but it goes thru the sequence one time and then stops for about 2 minutes, how can I get the sequence to run continuously?

BNNorman, 2017-11-29:
There isn't enough information here to help or reproduce the problem. 

1. I'm assuming that it starts again after the 2 minute pause.
2. Which arduino are you running it on?
3. Presumably you have altered rgbstripsequence.ino to match the parameters of your strip - try posting it for people to look at.
4. Do the symptoms change if you change the animation sequence? (add/remove entries)
5. Do you still have an ALA_ENDSEQ at the end of the list? The code uses this to work out the size of the list - I don't know what it would do if it's missing but maybe it would think the list is longer and disappear into neverland for a while (2 minutes?)
6. Have you tried adding Serial.Print statements to see where it gets to?
7. Line 159 in AlaLedRgb.cpp is where the code selects the next animation from the sequence - you could try printing the value of AnimSeqLen - just to make sure it matches the number of animations in the list.

I've run out of thoughts now. Good luck.

Cue777, 2017-11-30:
I am using a Arduino Uno R3. I ran the sequence just as it was downloaded. I dont know what serial.print statements is?

I do not have a line 159 or anything that says  AlaRgb.cpp in the code? I see that file in the folder that I downloaded though but don't know how to open it

If I remove the bubbles and bouncing balls line it seems to run continuously.
I am new to this.

BNNorman, 2017-11-30:
It's line 159 in AlaRgb.cpp which is part of the library. You should find it in libraries\ALA\src. A CPP file is a C++ source file and can be opened with any text editor. Also, the arduino .ino file is also text - just a C++ file with a  different extension.

Serial.print() can be used to output messages to the Tools->Serial Monitor window whilst your code is running and can help debugging by printing out variable values and messages to check that the code routes where you expect it to. That's another story so let's focus on what you just said.

Since you have isolated the problem to the bubbles and bouncing balls animations let me ask if you have correctly configured the RgbStripSequence sketch for your LED strip.

This line in particluar in the setup() section:-

rgbStrip.initWS2812(60, 6);

Are you using WS2812 LEDs and do you have 60 of them? The number 6 refers to the GPIO pin that drives the LEDs (The data line which runs from one LED to the next in a daisy chain).

The animations would run but if you tell the code you have more LEDS than you actually have it'll be busy clocking the extra LEDs and you won't see anything - maybe for the 2 minutes you mentioned? You can tell the code you have less LEDs - it just wont use the ones at the end of a strip/matrix so you might see a blank section.

So, how many LEDs are in the strip/matrix you are using?

I must say that when I was playing with ALA I found it quite tight on memory - the UNO doesn't have a lot of RAM. It holds a memory buffer for the LEDS at 3 bytes per LED (RGB) so, with 2K SRAM you won't be able to drive more than 2000/3=666 leds. In fact you won't drive anywhere near that since the program needs space for a stack and space for variables.

Since you are a newbie I don't mind helping, if I can - we were all there once.

Cue777, 2017-11-30:
Thanks for the info, I am using 200 WS2811 leds and have changed that number in the program before loading it. I don't need the last 2 animations so I am happy with the way it is working now.

BNNorman, 2017-11-30:
Ok, but I suggest, for peace of mind. You reduce the number in order to check bubbles etc works - you will have a blank section at the end of the LEDs which doesn't animate. I was pushing it with a 144 LED string using ALA.
When you compile your sketch the Arduino GUI tells you how much space the sketch takes and how much ram is left - best check that number. If you upgrade to a MEGA2560 you will get 4x as much RAM and your program is less likely to disappear into neverland.

BNNorman, 2017-12-01:
BTW - you might be interested in this 32kRAM, 48Mhz, 256Kb rom only £10:-

https://www.indiegogo.com/projects/exen-mini-smallest-32bit-arduino-comp-dev-board-diy#/