bportaluri/ALA #18: Freeze

dansleboby, 2018-01-29:
After some time of run 2-3 hours the animation freeze and I need to reset Arduino :(

bportaluri, 2018-02-20:
I have left the animations running for the entire night without any problem

dansleboby, 2018-02-20:
Here is my code
```c
#include "AlaLedRgb.h"

AlaLedRgb rgbStrip;

AlaSeq seq[] =
{
  { ALA_FIRE, 1000, 20000, alaPalFire },
  { ALA_SPARKLE, 5000, 15000, alaPalRainbow },
  { ALA_SPARKLE2, 5000, 15000, alaPalRainbow },
  { ALA_STROBO, 5000, 15000, alaPalRainbow },
  { ALA_CYCLECOLORS, 5000, 15000, alaPalRainbow },
  { ALA_PIXELSHIFTRIGHT, 5000, 15000, alaPalRainbow },
  { ALA_PIXELSHIFTLEFT, 5000, 15000, alaPalRainbow },
  { ALA_PIXELBOUNCE, 5000, 15000, alaPalRainbow },
  { ALA_PIXELSMOOTHSHIFTRIGHT, 5000, 15000, alaPalRainbow },
  { ALA_PIXELSMOOTHSHIFTLEFT, 5000, 15000, alaPalRainbow },
  { ALA_PIXELSMOOTHBOUNCE, 5000, 15000, alaPalRainbow },
  { ALA_COMET, 5000, 15000, alaPalRainbow },
  { ALA_COMETCOL, 5000, 15000, alaPalRainbow },
  { ALA_BARSHIFTRIGHT, 5000, 15000, alaPalRainbow },
  { ALA_BARSHIFTLEFT, 5000, 15000, alaPalRainbow },
  { ALA_MOVINGBARS, 5000, 15000, alaPalRainbow },
  { ALA_MOVINGGRADIENT, 5000, 15000, alaPalRainbow },
  { ALA_LARSONSCANNER, 5000, 15000, alaPalRainbow },
  { ALA_LARSONSCANNER2, 5000, 15000, alaPalRainbow },
  { ALA_FADEIN, 5000, 15000, alaPalRainbow },
  { ALA_FADEOUT, 5000, 15000, alaPalRainbow },
  { ALA_FADEINOUT, 5000, 15000, alaPalRainbow },
  { ALA_GLOW, 5000, 15000, alaPalRainbow },
  { ALA_FADECOLORS, 5000, 15000, alaPalRainbow },
  { ALA_FADECOLORSLOOP, 5000, 15000, alaPalRainbow },
  { ALA_PIXELSFADECOLORS, 5000, 15000, alaPalRainbow },
  { ALA_BOUNCINGBALLS, 5000, 15000, alaPalRainbow },
  { ALA_BUBBLES, 5000, 15000, alaPalRainbow },
  { ALA_ENDSEQ }
};

void setup()
{
  delay(1000);
  
  rgbStrip.initWS2812(60, 6);
  
  rgbStrip.setBrightness(0xFFFFFF);

  rgbStrip.setAnimation(seq);
}

void loop()
{
  rgbStrip.runAnimation();
}
```

BNNorman, 2018-02-21:
It might help bportaluri if you provided more information such as which arduino you are running this on, how you are powering the arduino ( in case its a psu problem) and the output in the message pane of the arduino GUI when you upload your program. Also have you tried a different pin in case there's a problem with your hardware?

dansleboby, 2018-02-21:
I have an Arduino Uno, I try an other arduino Uno same bug, I try to change pin same
Upload details:
```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 5.0
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\gilbe\AppData\Local\Temp\arduino_build_328341/LED.ino.hex"
avrdude: writing flash (20324 bytes):

Writing | ################################################## | 100% 3.06s

avrdude: 20324 bytes of flash written
avrdude: verifying flash memory against C:\Users\gilbe\AppData\Local\Temp\arduino_build_328341/LED.ino.hex:
avrdude: load data flash data from input file C:\Users\gilbe\AppData\Local\Temp\arduino_build_328341/LED.ino.hex:
avrdude: input file C:\Users\gilbe\AppData\Local\Temp\arduino_build_328341/LED.ino.hex contains 20324 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.38s

avrdude: verifying ...
avrdude: 20324 bytes of flash verified

avrdude done.  Thank you.
```

BNNorman, 2018-02-21:
Ah, that's better. Ok, I don't have a UNO but I have a 328P Nano (same memory, speed etc).  It's busy logging my boiler temperatures at the moment (got a problem)  but I'll put the program on it and see what happens.
One thought. Shouldn't be an issue I think, but what happens if you reduce the number of LEDs? More LEDs=more memory (3 bytes per) and some for animations (I think Fire builds an array IIRC )- possibly it's just running out of memory. The arduino's heap starts from one end of ram and the program at the other. If they meet in the middle then all hell could break loose. Similar check might be to reduce the number of animations. 
It's been a while since I visited this code. if you must run all the animations then maybe you need an arduino with more memory (MEGA?).
I'll let you know how I get on, could be a while, but I'd like you to consider the points I've made and see if changing anything solves the problem. 
Regards
Brian

dansleboby, 2018-02-21:
I think you right, I think it a memory issues, but I can loop all animation for 2-3 hours, I think we having some memory issues in the code itself

BNNorman, 2018-02-21:
Hmm, found a pro mini (same specs ar nano/uno) in one of my boxes. Ran your code and it hung within 1 hour. May have hung sooner but I added a 1 second delay to blink LED13 (which stopped blinking).
I noticed that the compiled code was 21898 bytes. The ardino uses 2K of it's 32K for itself leaving roughly 8102 bytes free for the heap and arrays. I think you are pushing it. Further investigation I'll leave to you but I would halve the number of animations run it. If ok, put half those taken out back until you find a limit  or buy a MEGA etc STM32F2's run at least 5x faster, have more ram and, in bulk from china I got 10 for about £15 - just a pain getting them to work with USB.

Memory issues - yeah, I'd bet my bottom dollar on it if I was American LOL.

BNNorman, 2018-02-21:
Took longer to hang the second time. But it hung. What's needed is feedback on what's happening to memory but it certainly seems to me like a memory issue. Have you tried to pin down which animation causes it?
As I mentioned - FIRE and BUBBLES allocate arrays when in use. At a quick glance, BUBBLES always allocates a new array whereas fire only allocates one if it doesn't exist (If I understand the code properly).
I would suggest commenting out bubbles and giving it another try.
Just trying that myself now - off to visit a bottle of wine and the TV whilst it runs. It is 7:40pm in the UK right now.

dansleboby, 2018-02-21:
I comment dire and bubbles I will make a try too

BNNorman, 2018-02-21:
Almost certainly Bubbles. It uses new() but never delete() so it gradually gobbles up memory.

BNNorman, 2018-02-21:
Mine is still banging on without bubbles. Clearly the cause. I'll leave that to bportaluri to fix.

dansleboby, 2018-02-21:
Yes me too no issues 2 hours of run without bubble and fire

BNNorman, 2018-02-21:
Definitely bubbles - still running and that's all I've left out.

BNNorman, 2018-02-22:
Has been running since 6:30 this morning - now 10:00 - so definately bubbles and new()

BNNorman, 2018-02-22:
Humble pie time - maybe not BUT I've been monitoring freeMemory(). With Just FIRE and BUBBLES running I get 1046 bytes free which dips briefly to 978 then goes back to 1046.
Bubbles on it's own stays at 1058, fire on its own stays at 1064 so that's odd.
With all animations running freeMemory() reports only 617 bytes free with no sign of dipping. Of course this could be partly affected by the calls to Serial.print to get the info out.
I'll leave it running and see if it stops again.

BNNorman, 2018-02-22:
Further info: After some time the 617 dipped to 549 (68 bytes) same size drop as that with fire and bubbles alone, and returned to 617. Well, it's been going for over two hours with all animations included. Weird!

BNNorman, 2018-02-22:
Well, I'm stumped now. Classic memory problem but, even after adding the extra coding to monitor free memory and spit out messages (which showed no changes except those mentioned above) the little so and so is still running.

bportaluri, 2018-08-11:
Hi all the memory management is almost the same for fire/bubbles/balls animations.
Arrays are allocated at first run in the animation function based on pxPos variable and deleted when animation changes in the set animation.
I'm 99% it is correct and there is no memory leak.
By the way, it can be related to memory fragmentation due to allocation/de-allocation of arrays. Driving 60 RGB pixels on a Arduino Uno is very close to the board memory limit. You can try with a Mega board. This will surely solve all your memory problems.