bportaluri/ALA #22: SAMD21 Cortex-M0+ Board Support Integration?

eeik, 2018-03-23:
Greetings, 🏞️

I've run into a problem on a project I was hoping to finish up shortly that involves the use of the **ALA Library**. As my feature list grew and supporting hardware came together, the libraries required to power the solution no longer fit within the SRAM on the Arduino Nano. I've now upgraded to the **MKR Zero** instead which boasts a luxurious expansion of SRAM to support all sorts of _fun animations_ and _palettes_ within the **ALA Library**! Sweet! Right!?

Well, as I try to re-compile my sketches, all of my other libraries port over just fine, but within the ALA Library, I'm seeing the dreaded do-not-pass-go before my eyes. Ugh! 👀

[image]

## I want to help! ⛏️

### SAMD21 Cortex-M0+ Board Background
Links to the SAMD core (MKR Zero variant) [documentation](https://www.arduino.cc/en/Guide/ArduinoMKRZero) and [source](https://github.com/arduino/ArduinoCore-samd/tree/master/variants/mkrzero) can be found here.

The pinout description lives within the ["variant.h"](https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrzero/variant.cpp) file within the source tree.

The more general support info can be found [here (store.arduino.cc page)](https://store.arduino.cc/usa/arduino-mkrzero)

### Initial ALA compilation issue
I've tracked down the initial compilation error to [line 114 of the _"chip_includes.h"_](https://github.com/bportaluri/ALA/blob/8c107265c356db1fb041f299cbd620316803d2d4/src/pinouts/chip_includes.h#L114) file. The **"#if defined"** commands are doing their job and trapping me within the **[=[-[ #error "Unknown Chip!" ]-]=]** state.

I see within this area of the ALA code that there are a few pinout mapping files supporting some of the most prominent chipsets for various Arduino packages. There seem to be minor variations within these files to support the various chip constraints, but I can't specifically attack this without a considerable amount of reverse-engineering and guesswork that I'm hoping someone might be able to help guide me on.

Is there some place within the documentation, **or can I help write some?**, that might help me understand specifically how to encode a new hypothetical "SAMD_MKR_Zero.h" pinout file to test out whether or not the ALA Library can be easily ported?

Sadly, my initial attempts have thus far been in vain to try and mimic the functional mapping files to the SAMD platform in order to tickle the right configuration parameters and get through compilation.

[image]

I presume it is also a safe bet that, once through these initial pin mappings, there may also be some some additional (deeper under the hood) incompatibilities that I'd need to then start tracking down after taking my first best guess at mimicking the other pinout diagrams.

**A few questions come to mind when considering trying to contribute here or instead seeking out an alternative library to re-learn and re-integrate ... please advise:**

1. Any pointers on how specifically I might start in crafting the requisite ".h" files to support this SAMD platform?
2. Do you have a checklist of some sort that you might use or be able to share with contributors as to what to be aware of when navigating the codebase?
3. Perhaps there's a few noteworthy documentation breadcrumbs within the code that I've not yet uncovered that might help guide me through?
4. Perhaps you might also have some sort of visualization or tech-spec drawings that you might be able to share that might logically arrange the functional components and notable hardware dependencies that the ALA Library relies upon?
5. In addition to the ALA specific code, what other logical hierarchies are involved and how entangled with the board pinout mapping files might they be? For instance, the AdaFruit library integration points seem like they shouldn't be show-stoppers in supporting a new board, but perhaps I'm mistaken?

Any info that might be worthwhile in hacking together a bridge to the SAMD21 board would be appreciated!

🌟 Many thanks! 🌟

eeik, 2018-03-23:
I am seeing that the [Adafruit NeoPixel Library](https://github.com/adafruit/Adafruit_NeoPixel) does support the **ATSAMD21 (Arduino Zero/M0 and other SAMD21 boards) @ 48 MHz**, so I'm hoping that means the ALA library can be tweaked with relative ease.

CaptClaude, 2018-04-16:
At the risk of stepping on @bportaluri 's toes, the ALA is currently based on an older version of the Adafruit NeoPixel library and he is hard at work updating it to the newest, which you say doesn't support ATSAMD21 boards.  I don't want divert people away from this awesome project, but there is a library called "FastLED" that does have support for those boards.  There is a very active Google+ group for support.  Start here: http://fastled.io/

bportaluri, 2018-08-11:
Hi the .h files in the pinouts directory are belonging to the TLC5940.
I have updated the NeoPixel library in my last commit.

cnstjohn, 2018-09-20:
@bportaluri Super excited to put your library to work but sorry to report that I pulled the latest and compiled for SAMD21 (Adafruit Feather M0 bluefruit specifically) but I still ran into the same compile issue as detailed by @eeik.   Like to know how to help.

coneyrabbit, 2020-02-22:
Any update on this? Still getting unknown chip for MKRZERO