bportaluri/ALA #24: Execute once?

rmccaskill, 2018-07-23:
I'd like to use this library to animate some LEDs for a 'power up' sequence and a 'power down' sequence of an installation I'm doing.

Is there a way to:
 - Run a sequence once
 - Then continue other code once the sequence is complete.

Example
 - "power up" sequence.
 - "Steady" sequence (until a button is pressed)
 - "power down sequence

rmccaskill, 2018-07-24:
I tried wrapping each animation sequence in a blocking timed while loop that would:

1) Set the animation
2) Calculate the duration of the sequence (using the code that setAnimation uses)
3) Run a loop for that duration calling .runAnimation inside the loop

It almost works, but the animation always 'over executes' and starts to loop again. I've debugged the animation duration vs the loop timing and it appears perfect... but the loop always starts to execute again and the lights star the sequence over again for a little bit

Example:

```
void playBlockingAnimation(AlaSeq seq_animation[])
{
  leds.setAnimation(seq_animation); //Sequence is 4 seconds.
  long animationDuration = getDuration(seq_animation);
  Serial.print("Animation Duration: "); Serial.println(animationDuration);

  long startT = millis();
  for( long timeStart = millis();  (millis()-timeStart) < animationDuration;  )
  {
    leds.runAnimation();
  }

  long actionDuration = millis() - startT;
  Serial.print("Actual Duration: "); Serial.println(actionDuration);
}

Output:
Animation Duration: 6500
Actual Duration: 6500
```

BNNorman, 2018-07-24:
I think you would be better creating an ALA_STOP animation , in AlaLedRgb.cpp, which sets a global flag to indicate the animation sequence has, well, stopped.
Since runAnimation only returns true (refresh done) or false (if next refresh is not due, or no such animation) you would also need to add methods to clear/inspect the stopped status. E.g.

```
leds.clearStoppedFlag();
while(1)
{
leds.runAnimation();
if (leds.getStoppedFlag()) break;
}
```

rmccaskill, 2018-07-25:
Thanks for the reply! REALLY great library here, by the way! I'm hoping to use it to add some 'personality' to a laser tripwire maze I'm building for a local attraction.

I think I'll take a look at that suggestion. Thanks.

BNNorman, 2018-07-25:
That sounds like fun.
BTW, I produced a modified version of the ALA library a while back which allows you to logically split a LED strip and run separate animations on the separate sections see https://github.com/BNNorman/Arduino-ALA. It also handles multiple strips on seperate pins - however, it is memory hungry and doesn't have a STOP option.

BNNorman, 2018-07-27:
Each instantiation of AlaLedRgb object has it's own private variables and an array for the LEDs so I guess they would be remembered. It should be possible to call leds.AlaLedRgb() to reset them followed by leds.initWS2812() (or other init) before running a new animation - that would be akin to creating a new object instance without using more memory.

I'm not a 'C' guy either - I have coded in C as a hobby in the past but there's still a lot for me to learn.

BNNorman, 2018-07-27:
I have managed to get this to work - restarting the sequence from the beginning. I suspect the sequence (seq) could be changed in loop().

In AlaLedRgb() I just copied the AlaLedRgb:AlaLedRgb() method and renamed it as reset(). C doesn't like the constructor being called directly.

My sketch loop looks like this:-

void loop()
{
  rgbStrip.reset();
  rgbStrip.setAnimation(seq);
  rgbStrip.clearStoppedFlag();
  while(1)
  {
  rgbStrip.runAnimation();
  if (rgbStrip.getStoppedFlag()) break;
  }
  Serial.println("Animation stopped");
  delay(5000);
}

Here's my serial monitor  output (debug messages placed at start of setAnimationFunc()). NOTE: Animation 504 is my ALA_STOP animation which just sets stoppedFlag to true.

setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)
setAnimationFunc(151)
setAnimationFunc(102)
setAnimationFunc(352)
setAnimationFunc(106)
setAnimationFunc(102)
setAnimationFunc(106)
setAnimationFunc(211)
setAnimationFunc(102)
setAnimationFunc(241)
setAnimationFunc(221)
setAnimationFunc(222)
setAnimationFunc(102)
setAnimationFunc(304)
setAnimationFunc(102)
setAnimationFunc(501)
setAnimationFunc(102)
setAnimationFunc(502)
setAnimationFunc(102)
setAnimationFunc(503)
setAnimationFunc(504)
Animation stopped
setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)
setAnimationFunc(151)
setAnimationFunc(102)
setAnimationFunc(352)
setAnimationFunc(106)
setAnimationFunc(102)
setAnimationFunc(106)
setAnimationFunc(211)
setAnimationFunc(102)
setAnimationFunc(241)
setAnimationFunc(221)
setAnimationFunc(222)
setAnimationFunc(102)
setAnimationFunc(304)
setAnimationFunc(102)
setAnimationFunc(501)
setAnimationFunc(102)
setAnimationFunc(502)
setAnimationFunc(102)
setAnimationFunc(503)
setAnimationFunc(504)
Animation stopped
setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)
setAnimationFunc(151)
setAnimationFunc(102)
setAnimationFunc(352)
setAnimationFunc(106)
setAnimationFunc(102)
setAnimationFunc(106)
setAnimationFunc(211)
setAnimationFunc(102)
setAnimationFunc(241)
setAnimationFunc(221)
setAnimationFunc(222)
setAnimationFunc(102)
setAnimationFunc(304)
setAnimationFunc(102)
setAnimationFunc(501)
setAnimationFunc(102)
setAnimationFunc(502)
setAnimationFunc(102)
setAnimationFunc(503)
setAnimationFunc(504)
Animation stopped
setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)
setAnimationFunc(151)
setAnimationFunc(102)
setAnimationFunc(352)
setAnimationFunc(106)
setAnimationFunc(102)
setAnimationFunc(106)
setAnimationFunc(211)
setAnimationFunc(102)
setAnimationFunc(241)
setAnimationFunc(221)
setAnimationFunc(222)
setAnimationFunc(102)
setAnimationFunc(304)

rmccaskill, 2018-07-29:
I'm using AlaLed, not AlaLedRGB as each laser is essentially one 'normal' LED. So, I'll give this a try there.  I did try creating a new instance of the library every time, but that didn't seem to work. Something was still remembered.

Thanks!

rmccaskill, 2018-07-29:
Can you try to make 3 different 'sequences'. Play seq1, then seq2, then seq3 THEN loop? Things get... messed up then.

BNNorman, 2018-07-29:
Source code looks very similar. I only have ws2812 to test with. I can't see why it won't work. Will have a go tomorrow.

BNNorman, 2018-07-30:
Well, it works ok for me.... with my WS2812 matrix on a NANO.....

Animation sequences starting.

Starting sequence 1
setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)
setAnimationFunc(151)
setAnimationFunc(102)
setAnimationFunc(504)
Sequence 1 stopped.
Pausing for 2 seconds.
Starting sequence 2
setAnimationFunc(352)
setAnimationFunc(106)
setAnimationFunc(102)
setAnimationFunc(106)
setAnimationFunc(211)
setAnimationFunc(102)
setAnimationFunc(241)
setAnimationFunc(221)
setAnimationFunc(222)
setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)
setAnimationFunc(151)
setAnimationFunc(102)
setAnimationFunc(504)
Sequence 2 stopped.
Pausing for 2 seconds.
Starting sequence 3
setAnimationFunc(304)
setAnimationFunc(102)
setAnimationFunc(501)
setAnimationFunc(102)
setAnimationFunc(502)
setAnimationFunc(102)
setAnimationFunc(503)
setAnimationFunc(504)
Sequence 3 stopped.
Pausing for 2 seconds.
Starting sequence 1
setAnimationFunc(102)
setAnimationFunc(101)
setAnimationFunc(105)

It is still banging away as I type.
I have attached my modified ALA library (unzip and replace yours). Also attached is my test sketch. I have modded AlaLed the same way as AlaLedRgb so you should be able to just adapt my sketch.

I don't have a laser to test with so if you still have problems I think I would need to know your complete hardware setup.

[ALA_NANO_RgbStripSequence.zip](https://github.com/bportaluri/ALA/files/2239943/ALA_NANO_RgbStripSequence.zip)
[ALA.zip](https://github.com/bportaluri/ALA/files/2239948/ALA.zip)

rmccaskill, 2018-07-30:
I'm not using an RBG strip - Just AlaLed but I'll take a look. Thanks!

rmccaskill, 2018-07-31:
Sorry to keep taking our time, but I do appreciate it! 

I think I'm going crazy.... I've spent hours on this (more than I care to say!) as I love the idea of being able to know my PC will be playing a 'power up sound' that lasts 3 seconds and I can use your library to time a powerup lighting sequence to match the exact timing.  The result would be great!

I see that you made the changes in AlaLED also so I changed my code to your reserved words, functions, etc in your library and I still can't get it to work.  I'll zip up my ino file for your to take a look (if you're interested).

Maybe I'll change our my Leonardo for another board and see if that makes a difference?

I have some questions about your implementation:

1) Won't the ALA_STOP take 1 second to run? 
2) You have an ALA_STOP in seq1 and seq3 but NOT seq2. Why not?
3) Shouldn't there be an ALA_ENDSEQ at the end of each sequence?

Thanks again.

Note: Locally, I added debugging code to your setAnimation() to write out the sequence to run. I also added debugging to setAnimationFunc() to track each time a function was called. They don't match after the first loop (which is what I was seeing as well):

```
Starting sequence 1
Sequence to run
102
101
105
151
102
504
-End. Size: 6 Duration: 21000
setAnimationFunc: 105
setAnimationFunc: 151
setAnimationFunc: 102
setAnimationFunc: 504
Setting stoppedFlag
Sequence 1 stopped.
Pausing for 2 seconds.
```
What happened to 102 and 101? 
ACK!
[LightTest_Rick.zip](https://github.com/bportaluri/ALA/files/2243322/LightTest_Rick.zip)

BNNorman, 2018-07-31:
Q1:
Missing ALA_STOP - whoops, just a mistake. However, I left the animation running for quite some time on my PC so it would have executed numerous ALA_STOP calls.

Q2:
ALA_STOP only runs for 1 second if the call to runAnimation() continues. In my while(1) loop the runAnimation() call is immediately followed by if (getStoppedFlag()) break; which terminates the while loop on the first tick of ALA_STOP. 

Q3:
ALA_ENDSEQ - another oversight BUT the ALA code only uses that to determine the size of the sequence array. Nothing more.

I have corrected my sketch and it still runs correctly (using AlaLedRgb). Thanks for pointing those errors out.

Hmm, I have compiled your LightTest sketch (AlaLed) and run it on a NANO and WOWZA it doesn't work correctly. Therefore there must be some difference between AlaLed and AlaLedRgb. 
```

Power Up
----------------------
setAnimationFunc(106)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

Power Drown
----------------------
setAnimationFunc(103)
setAnimationFunc(302)
setAnimationFunc(504)
----------------------

Power Up
----------------------
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

Power Drown
----------------------
----------------------

Power Up
----------------------
setAnimationFunc(221)
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

Power Drown
----------------------
setAnimationFunc(302)
setAnimationFunc(504)
----------------------

Power Up
----------------------
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

Power Drown
----------------------
----------------------
```

It appears to be screwing up the next animation sequence - even after a reset.

I'll take a look at the ALA code again - the same bug may exist in my GitHub modified version.

BNNorman, 2018-07-31:
I think I've fixed it.....

```
Power Up
----------------------
setAnimationFunc(106)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

waiting 5 seconds
Power Down
----------------------
setAnimationFunc(102)
setAnimationFunc(103)
setAnimationFunc(302)
setAnimationFunc(504)
----------------------

Power Up
----------------------
setAnimationFunc(106)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

waiting 5 seconds
Power Down
----------------------
setAnimationFunc(102)
setAnimationFunc(103)
setAnimationFunc(302)
setAnimationFunc(504)
----------------------

Power Up
----------------------
setAnimationFunc(106)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

waiting 5 seconds
Power Down
----------------------
setAnimationFunc(102)
setAnimationFunc(103)
setAnimationFunc(302)
setAnimationFunc(504)
----------------------

Power Up
----------------------
setAnimationFunc(106)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(221)
setAnimationFunc(301)
setAnimationFunc(504)
----------------------

waiting 5 seconds
```
As you can see the animation sequences are always the same. I don't have the same hardware as you so I can only go on Serial Monitor messages.

Basically the problem was in  runAnimation() and setAnimation() methods. The animStartTime was being reset in the wrong place AND in runAnimation() the check for the end of an animation was not working.

However, you should run the code for a long time - the millis() function will eventually wrap around after 59 days I think.  Anyway, a frequent restart should avoid unpleasantries.

Took ages to pin down.

Give it a go and let me know how you get on.

[AlaLed.zip](https://github.com/bportaluri/ALA/files/2245217/AlaLed.zip)

rmccaskill, 2018-08-01:
Running now and it looks perfect!  

Now I need to go through your code and see the differences to understand how you fixed it. I spent hours trying to figure it out!

Any plans to merge this into the main codebase?

BNNorman, 2018-08-01:
I spent hours trying to figure it out too. LOL.

To my mind the AlaLed code should have looked  like the AlaLedRgb code (which worked straight away) but without a palette. AlaLed also looked unfinished because in runAnimation() it was using millis() several times as the current  time whereas AlaLedRgb called it once and assigned it's value to a variable cTime which meant the timing code wasn't reading a different value each time it looped. But that would have a minor side effect. 

AlaLedRgb also uses an animSeqStartTime variable which is missing from AlaLed. I think adding that would have fixed the code more easily (on reflection).

In runAnimation() the original code was calculating the next animation in the sequence based on the time the individual animation started (animStartTime) and it was coming up with the wrong answer. I changed it so that it worked on when the sequence started.

I wouldn't merge this AlaLed with the main codebase because I think it can be tidied up so that it is more like AlaLedRgb - I would need to rework it. Also, the Serial.print() calls need removing.

I'll think about cleaning up my fixes and offering the code up.

Last comment. AlaLedRgb includes code for PWM - just like AlaLed - did you try using that? If that works then Alaled would not be needed at all.

rmccaskill, 2018-08-02:
I didn't try using that... I just made an assumption that it wouldn't work :(

I also noticed that the docs mentioned that the larson scanners weren't available for LED (they looked like rbgled only) but they code is there, I tested and they work great.

Thank so SO much for your help! Our event is this weekend. I'll send some pics after the event.

BNNorman, 2018-08-02:
Yeah, would like to see that.

I have downloaded the latest ALA-master and added the required MODs. The MODs are more in line with AlaLedRgb and, therefore, more likely to be accepted by bportaluri.

In particular AlaLed.cpp now uses the animSeqStartTime variable and runAnimation() uses the cTime variable instead of multiple calls to millis() - exactly the same way as AlaLedRgb.cpp does it.

I have also redefined ALA_STOP as 999 - just to leave a gap should bportaluri want to add more animations.

I will attempt to clone ALA and try to get it merged into the main branch - but I'm unfamiliar with using github in that way - basically I just use it to store code from my little projects in case anyone else finds it useful. I'll be using the Git Gui for windows with my Python coding GIT is built into the PyCharm GUI.

I have read your github page for this. Looks interesting.

[ALA-master.zip](https://github.com/bportaluri/ALA/files/2252330/ALA-master.zip)

bportaluri, 2018-08-05:
Hi all. I'm the author of the library. Thank you for your feedback. I think you have spotted some problems.
I agree about changes related to the AlaLed/AlaLedRgb code related to animSeqStartTime. I will merge them to the master branch.
However I do not agree about the 'stop' feature. The entire ALA library is based on a non-blocking design and the proposed changes are not in line with this approach.
I'm working on a more elegant solution that I will merge in the following days. I will include e new example sketch that should satisfy your needs.

BNNorman, 2018-08-06:
I look forward to seeing your alternative to a STOP animation.

rmccaskill, 2018-08-08:
@bportaluri a fellow IBMer? Nice to meet you. Thanks for the great library!

bportaluri, 2018-08-08:
Left IBM one year ago...

bportaluri, 2018-08-11:
Hi guys. I have just committed my implementation of the STOP animation function.
I have also fixed AlaLed timing issues.
Please give it a try and let me know so i can release the new library version.
Thank you all for the great feedback.

BNNorman, 2018-08-12:
forget that - I expected the folder timestamps to have changed.